abobija/esp-idf-rc522 #2: Support I2C

michaelwittig, 2020-05-17:
I wonder if it is possible to support both SPI and I2C.

I try to connect a ESP32-LyraT V4.3 which does not support SPI.

abobija, 2020-09-29:
Currently only SPI is supported.
I will try to support I2C as well, soon.

michaelwittig, 2020-09-29:
I can post my port here

rc522.c:

```c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "i2c_bus.h"
#include "board.h"

#include "rc522.h"

static i2c_bus_handle_t i2c_handle;

static int i2c_addr = (0x28 << 1) | I2C_MASTER_WRITE;

esp_err_t rc522_write_n(uint8_t reg, uint8_t n, uint8_t *data) {
    return i2c_bus_write_bytes(i2c_handle, i2c_addr, &reg, sizeof(reg), data, n);
}

esp_err_t rc522_write(uint8_t reg, uint8_t val) {
    return rc522_write_n(reg, 1, &val);
}

uint8_t rc522_read(uint8_t reg) {
    uint8_t data = 0xFF;
    ESP_ERROR_CHECK(i2c_bus_read_bytes(i2c_handle, i2c_addr, &reg, sizeof(reg), &data, 1));
    return data;
}

esp_err_t rc522_set_bitmask(uint8_t reg, uint8_t mask) {
    return rc522_write(reg, rc522_read(reg) | mask);
}

esp_err_t rc522_clear_bitmask(uint8_t reg, uint8_t mask) {
    return rc522_write(reg, rc522_read(reg) & ~mask);
}

esp_err_t rc522_antenna_on() {
    esp_err_t ret = ESP_OK;

    if(~ (rc522_read(0x14) & 0x03)) {
        ret = rc522_set_bitmask(0x14, 0x03);
        if(ret != ESP_OK) {
            return ret;
        }
    }

    return rc522_write(0x26, 0x60); // 43dB gain
}

esp_err_t rc522_init() {
    esp_err_t ret = ESP_OK;

    i2c_config_t i2c_cfg = {
        .mode = I2C_MODE_MASTER,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000,
    };
    ret = get_i2c_pins(I2C_NUM_0, &i2c_cfg);
    if (ret != ESP_OK) {
        return ret;
    }
    i2c_handle = i2c_bus_create(I2C_NUM_0, &i2c_cfg);

    // ---------- RW test ------------
    ret = rc522_write(0x24, 0x25);
[... 189 lines cut ...]
        free(res_data);

        result = rc522_anticoll(&res_data_n);

        if(result != NULL) {
            uint8_t buf[] = { 0x50, 0x00, 0x00, 0x00 };
            uint8_t* crc = rc522_calculate_crc(buf, 2);

            buf[2] = crc[0];
            buf[3] = crc[1];

            free(crc);

            res_data = rc522_card_write(0x0C, buf, 4, &res_data_n);
            free(res_data);

            rc522_clear_bitmask(0x08, 0x08);

            return result;
        }
    }

    return NULL;
}
```

abobija, 2020-09-29:
So, you managed to adapt code to communicate via I2C. I guess it works for you.
Nice. This can be reference for me.
Thanks!

michaelwittig, 2020-09-29:
Yes, it works. And I only had to make very few changes.

Nodraak, 2021-03-24:
It was not obvious to me, so quick note for future readers:

It seems that the RC522 is hardwired to use SPI, therefore using I2C (or UART) requires a (not so easy, in the case if I2C) physical modification of the pcb to cut one or two copper trace(s): https://forum.arduino.cc/index.php?topic=442750.0

abobija, 2021-03-24:
Hi @Nodraak 
Yes, that's true. I have ordered this one https://ebay.to/3ceqSZr which is hardwired for I2C.
When it arrives I will modify this library to support both, SPI and I2C.

PR #11 (2022-08-26): Version 2.0.0
This resolves #2, resolves #5, closes #8, resolves #9.

## What's new in this version?

- Instead of passing pointer to handler function via the config structure, this version uses event loops which is the core feature of esp-idf framework.
- Besides default SPI protocol, library now supports I2C as well. Thanks to @michaelwittig for posting his code in #2
- Library now use SPI's FULLDUPLEX by default, and can be changed to HALFDUPLEX via the configuration structure. Thanks to @Archy404 for posting his tweak code in #8
- Library can be now used on already initialized SPI bus, by  [... cut]
closed 2022-08-26 by abobija