abobija/esp-idf-rc522 #4: rc522_anticoll() has no arguments

Jaszuro, 2021-01-28:
Hi!

I'm trying to modify your code to be able to handle different lengths of IDs, and found this line:
https://github.com/abobija/esp-idf-rc522/blob/826bbb2864b3017a95b6d001fd82fb5a8844a48f/rc522.c#L287
But it seems that the called function has no arguments at all.

abobija, 2021-03-16:
Hi @Jaszuro,
sorry for very late response.

Yes, `rc522_anticoll` does not receive any arguments because that function always returns 5 bytes long array.
You are right for that line, I was mistakenly provide argument to `rc522_antcoll` function. I have fixed that.

To be honest with you, I don't make enough research about RFID protocol and cards so I was thinking that all cards have serial number with fixed length of 5 bytes.

Which version of card do you have and how long is its serial number?
And have you managed to adapt this library to work with different serial number lengths?

Regards,
Alija

Jaszuro, 2021-03-18:
Hi @abobija,

I'm no expert on RFID also. I just have the luck of having 2 types of cards. The Mifare Classic does provide 5 byte serial number (4+1CRC if i'm correct), but for example Mifare Ultralight (NTAG213?) provides more (7+?CRC). My Chinese clone RFID-RC522 board could not handle the latter even after the replacement of the weaker inductors, so i didn't manage to read with that, but my phone told me the more than 5 bytes and i was pursuing to recognize those too.

abobija, 2021-03-19:
Thanks for the very useful feedback @Jaszuro
Mine module is Chinese clone as well, hope I will succeed to manage it to work when I get different type of cards/tags.
I will definitely in future research more about RFID-RC522 module to make this library more flexible.

closed 2021-03-24 by abobija