abobija/esp-idf-rc522 #5: Why is device accessed using SPI_DEVICE_HALFDUPLEX?

adrien-n, 2021-07-09:
Hi,

First of all, thanks for your work on the library.

I'm trying to use an rc522 on the same bus as an sd card reader through sdspi. I therefore need a single bus configuration that works for both devices. Unfortunately, in practice, sdspi requires setting up the bus to use DMA as far as I can tell (or buffers will be very very small) and devices that use SPI_DEVICE_HALFDUPLEX are not compatible with that (it's slightly more complicated than that actually[1]).

I've tried to remove that flag but then my tags don't get recognized anymore. Initialization (including tests) succeeds. I've tried to understand why it was used but I've got absolutely no idea and I found no reference whatsoever.

Can you shine a light on this matter? Is there a specific reason/timing/reference for this?

Thanks!

[1] https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#spi-known-issues

wieserh, 2021-11-02:
Well I just adjusted the code to always use full duplex. For my use case there i no downside. There are not many modifications required. I did encounter some issues with heap corruption though, currently trying to find out whether I introduced them and how to fix these.

dahmadjid, 2021-12-27:
I have rn the same issue. im using a tft screen and sdcard together with the rfid. DMA is necessary for everything to work. @wieserh how easy was it to change everything to full duplex ? do you have a modified branch or any idea on what i should change on my side ? thanks in advance

PR #11 (2022-08-26): Version 2.0.0
This resolves #2, resolves #5, closes #8, resolves #9.

## What's new in this version?

- Instead of passing pointer to handler function via the config structure, this version uses event loops which is the core feature of esp-idf framework.
- Besides default SPI protocol, library now supports I2C as well. Thanks to @michaelwittig for posting his code in #2
- Library now use SPI's FULLDUPLEX by default, and can be changed to HALFDUPLEX via the configuration structure. Thanks to @Archy404 for posting his tweak code in #8
- Library can be now used on already initialized SPI bus, by  [... cut]
closed 2022-08-26 by abobija