abobija/esp-idf-rc522 #6: 提示SHA256校验失败 反复重启

cnlnlhb, 2022-01-07:
您好，我运行了例程，芯片使用的是ESP32-C3，报错SHA256校验失败，反复重启
错误日志：

[image]

cnlnlhb, 2022-01-08:
您好，为啥关闭了？

abobija, 2022-01-08:
@cnlnlhb do you speak english?

cnlnlhb, 2022-01-08:
@abobija 
Hello, sorry for using Chinese to ask the question earlier. I ran the routine, the chip uses ESP32-C3, and it reported an error that the SHA256 verification failed, and restarted repeatedly. The development environment and hardware do not have this problem when compiling other code.
The error log is as posted before.
Looking forward to your reply, thank you.

abobija, 2022-01-08:
Did you set target to C3 with the next command `idf.py set-target esp32c3` ?

If that does not resolve your problem it might be some sort of power issues, check this link:
https://www.esp32.com/viewtopic.php?f=13&t=22302

closed 2022-01-11 by abobija