abobija/esp-idf-rc522 #7: No such file or directory

easylady, 2022-01-10:
hello!
Prompt for file not found,The content is as follows:
../main/ESP32_LittleVGL_V8.c:17:10: fatal error: rc522.h: No such file or directory
 #include "rc522.h"

But I have made sure to include it

abobija, 2022-01-10:
Hi!
Are you sure that you clone this repository `esp-idf-rc522` into `components` folder of your project?

easylady, 2022-01-11:
Hello!
I used the following command to get the components to my "Components Folder"
Like this:git submodule add https://github.com/abobija/esp-idf-rc522.git components/esp-idf-rc522
and I can also find it under my "Components folder"

easylady, 2022-01-11:
Hello!
I use the command "ESP-IDF FULL CLEAN" to clear the compilation and recompile, and the problem is solved.
Thank you!

abobija, 2022-01-11:
Glad to hear that you've fix the problem!

closed 2022-01-11 by abobija