abobija/esp-idf-rc522 #10: Problem while Calling again rc522_start(start_args) After rc522_destroy()

vigyanabikshu1, 2022-08-04:
#include <stdio.h>
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "rc522.h"

static const char* TAG = "RFID";

void rfidtag_handler(uint8_t* sn);

uint8_t rfidready=0;

const rc522_start_args_t start_args = {
        .miso_io  = RC522_MISO,
        .mosi_io  = RC522_MOSI,
        .sck_io   = RC522_SCK,
        .sda_io   = RC522_SDA,
        .callback = &rfidtag_handler,

        // Uncomment next line for attaching RC522 to SPI2 bus. Default is VSPI_HOST (SPI3)
        //.spi_host_id = HSPI_HOST
    };

void app_main(void)
{
    while(1){
      vTaskDelay(1000 / portTICK_PERIOD_MS);
      printf(".");
      fflush(stdout);
      
      if(rfidready==0){
        rc522_start(start_args);
        rfidready=1;
      }
    }
}

void rfidtag_handler(uint8_t* sn) { // serial number is always 5 bytes long
    ESP_LOGI(TAG, "Tag: %#x %#x %#x %#x %#x",
        sn[0], sn[1], sn[2], sn[3], sn[4]
    );
    
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    rc522_destroy();
    rfidready=0;
}

Log
entry 0x40080694
I (29) boot: ESP-IDF v4.4 2nd stage bootloader
I (29) boot: compile time 08:50:28
I (29) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09dc4h ( 40388) map
I (125) esp_image: segment 1: paddr=00019dec vaddr=3ffb0000 size=02494h (  9364) load
I (129) esp_image: segment 2: paddr=0001c288 vaddr=40080000 size=03d90h ( 15760) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=18ce0h (101600) map
I (177) esp_image: segment 4: paddr=00038d08 vaddr=40083d90 size=08acch ( 35532) load
[... 194 lines cut ...]
I (198) boot: Disabling RNG early entropy source...
I (211) cpu_start: Pro cpu up.
I (212) cpu_start: Starting app cpu, entry point is 0x400810a4
0x400810a4: call_start_cpu1 at /home/v-robotics/esp6/esp-idf/components/esp_system/port/cpu_start.c:156

I (198) cpu_start: App cpu up.
I (226) cpu_start: Pro cpu start user code
I (226) cpu_start: cpu freq: 160000000
I (226) cpu_start: Application information:
I (231) cpu_start: Project name:     main
I (235) cpu_start: App version:      1
I (240) cpu_start: Compile time:     Aug  4 2022 08:50:14
I (246) cpu_start: ELF file SHA256:  24761f48d297bde2...
I (252) cpu_start: ESP-IDF:          v4.4
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB2DB8 len 0002D248 (180 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (282) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (289) heap_init: At 4008C85C len 000137A4 (77 KiB): IRAM
I (296) spi_flash: detected chip: generic
I (300) spi_flash: flash io: dio
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (316) ESP-RC522: Initialized (firmware: 0x92)

abobija, 2022-08-04:
Function `rc522_destroy` should not be called inside of `rfidtag_handler` because it is in the same task. You can set some type of flag (e.g. bool variable) inside of `rfidtag_handler` and then in while loop inside of `app_main` check that flag and call `rc522_destroy` function.

vigyanabikshu1, 2022-08-04:
Than you very much for your kind support. Now solved the problem. I am still getting small problem after 4 or 5 scans.

Log
entry 0x40080694
I (29) boot: ESP-IDF v4.4 2nd stage bootloader
I (29) boot: compile time 08:50:28
I (29) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09e1ch ( 40476) map
I (126) esp_image: segment 1: paddr=00019e44 vaddr=3ffb0000 size=02494h (  9364) load
I (129) esp_image: segment 2: paddr=0001c2e0 vaddr=40080000 size=03d38h ( 15672) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=18dd8h (101848) map
I (177) esp_image: segment 4: paddr=00038e00 vaddr=40083d38 size=08b24h ( 35620) load
I (192) esp_image: segment 5: paddr=0004192c vaddr=50000000 size=00010h (    16) load
I (198) boot: Loaded app from partition at offset 0x10000
I (198) boot: Disabling RNG early entropy source...
I (212) cpu_start: Pro cpu up.
I (212) cpu_start: Starting app cpu, entry point is 0x400810a4
0x400810a4: call_start_cpu1 at /home/v-robotics/esp6/esp-idf/components/esp_system/port/cpu_start.c:156

I (198) cpu_start: App cpu up.
I (226) cpu_start: Pro cpu start user code
I (226) cpu_start: cpu freq: 160000000
I (226) cpu_start: Application information:
I (231) cpu_start: Project name:     main
I (235) cpu_start: App version:      1
I (240) cpu_start: Compile time:     Aug  4 2022 08:50:14
I (246) cpu_start: ELF file SHA256:  4458a0f0e19a6b54...
I (252) cpu_start: ESP-IDF:          v4.4
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB2DC0 len 0002D240 (180 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (282) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (289) heap_init: At 4008C85C len 000137A4 (77 KiB): IRAM
I (296) spi_flash: detected chip: generic
I (300) spi_flash: flash io: dio
I (305) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (317) ESP-RC522: Initialized (firmware: 0x92)
.........I (1337) RFID: Tag: 0x50 0x41 0xc5 0xd4 0
I (1347) RFID: Tag: 5041c5d40
.I (1347) RFID: RFID Destroy...
I (1347) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1357) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1357) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1367) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1377) RFID: RFID Start again...
I (1387) ESP-RC522: Initialized (firmware: 0x92)
........I (2257) RFID: Tag: 0x50 0x41 0xc5 0xd4 0
I (2267) RFID: Tag: 5041c5d40
.I (2327) RFID: RFID Destroy...
I (2327) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2327) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2337) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2347) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2357) RFID: RFID Start again...
I (2357) ESP-RC522: Initialized (firmware: 0x92)
........I (3247) RFID: Tag: 0x50 0x41 0xc5 0xd4 0
[... 17 lines cut ...]
..
assert failed: spi_device_transmit spi_master.c:865 (ret_trans == trans_desc)

Backtrace:0x400818de:0x3ffba1e00x400859e1:0x3ffba200 0x4008b541:0x3ffba220 0x400d80fd:0x3ffba340 0x400d4fa4:0x3ffba370 0x400d4fbf:0x3ffba3c0 0x400d52b0:0x3ffba3f0 0x400d530a:0x3ffba420 0x400d5374:0x3ffba450 0x4008859d:0x3ffba470 
0x400818de: panic_abort at /home/v-robotics/esp6/esp-idf/components/esp_system/panic.c:402

0x400859e1: esp_system_abort at /home/v-robotics/esp6/esp-idf/components/esp_system/esp_system.c:121

0x4008b541: __assert_func at /home/v-robotics/esp6/esp-idf/components/newlib/assert.c:85

0x400d80fd: spi_device_transmit at /home/v-robotics/esp6/esp-idf/components/driver/spi_master.c:865 (discriminator 1)

0x400d4fa4: rc522_write_n at /home/v-robotics/esp6/IoTAttendance/test/RFID/build/../main/rc522.c:93

0x400d4fbf: rc522_write at /home/v-robotics/esp6/IoTAttendance/test/RFID/build/../main/rc522.c:101

0x400d52b0: rc522_request at /home/v-robotics/esp6/IoTAttendance/test/RFID/build/../main/rc522.c:343

0x400d530a: rc522_get_tag at /home/v-robotics/esp6/IoTAttendance/test/RFID/build/../main/rc522.c:375

0x400d5374: rc522_task at /home/v-robotics/esp6/IoTAttendance/test/RFID/build/../main/rc522.c:454

0x4008859d: vPortTaskWrapper at /home/v-robotics/esp6/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 4458a0f0e19a6b54

closed 2022-08-04 by vigyanabikshu1

vigyanabikshu1, 2022-08-04:
Confirming taskterminated before free(hndl) solved the above issue.Thanks again abobija ji.

```
bool taskterminated=0;

void rc522_destroy() {
    if(! hndl) { return; }
    taskterminated=0;
    rc522_pause(); // stop timer
    hndl->running = false; // task will delete itself
  
    if(hndl->spi) {
        spi_bus_remove_device(hndl->spi);
        spi_bus_free(hndl->config->spi_host_id);
        hndl->spi = NULL;
    }

    free(hndl->config);
    hndl->config = NULL;

    while(!taskterminated);

    free(hndl);
    hndl = NULL;
}

static void rc522_task(void* arg) {
    while(hndl->running) {
        if(!hndl->scan_started) {
            vTaskDelay(100 / portTICK_PERIOD_MS);
            continue;
        }

        uint8_t* serial_no = rc522_get_tag();

        if(serial_no && ! hndl->tag_was_present_last_time) {
            rc522_tag_callback_t cb = hndl->config->callback;
            if(cb) { cb(serial_no); }
        }
        
        if((hndl->tag_was_present_last_time = (serial_no != NULL))) {
            free(serial_no);
            serial_no = NULL;
        }

        int delay_interval_ms = hndl->config->scan_interval_ms;

        if(hndl->tag_was_present_last_time) {
            delay_interval_ms *= 2; // extra scan-bursting prevention
        }

        vTaskDelay(delay_interval_ms / portTICK_PERIOD_MS);
    }
    taskterminated=1;
    vTaskDelete(NULL);
}
```

abobija, 2022-08-04:
Glad it helps. Good luck with your project.
Cheers.