abobija/esp-idf-rc522 #12: Error Event, Crashes The ESP32

parthbhat13, 2022-09-06:
Hey, 
Greetings, hope you are doing good. 
i really loved your library and i had it up and running like hardly in less than 5 minutes. really great job. 
perhaps, i purposely wanted to try and remove the module, and wanted the library to throw errors and halt as it is without it trying anymore, but apparently even if i checked the error states, it seems like your code is not really returning anything back. i wanted to sit and check it line by line, but i had a lot on my plate to solve it so rather thought to bug you instead. 

here's the code i used 
```
#include <esp_log.h>
#include <inttypes.h>
#include "RC522.h"

static const char* TAG = "rc522-demo";
static rc522_handle_t scanner;

static void rc522_handler(void* arg, esp_event_base_t base, int32_t event_id, void* event_data)
{
    rc522_event_data_t* data = (rc522_event_data_t*) event_data;

    switch(event_id) {
        case RC522_EVENT_TAG_SCANNED: {
                rc522_tag_t* tag = (rc522_tag_t*) data->ptr;
                ESP_LOGI(TAG, "Tag scanned (sn: %" PRIu64 ")", tag->serial_number);
            }
            break;
    }
}

void app_main()
{
    esp_err_t err;

    rc522_config_t config = {
        .spi.host = VSPI_HOST,
        .spi.miso_gpio = 19,
        .spi.mosi_gpio = 23,
        .spi.sck_gpio = 18,
        .spi.sda_gpio = 5,
    };

    err = rc522_create(&config, &scanner);
    if(err != ESP_OK)
        ESP_LOGE(TAG, "Failed to create RC522 scanner: %s", esp_err_to_name(err));

    
    err = rc522_register_events(scanner, RC522_EVENT_ANY, rc522_handler, NULL);
    if(err != ESP_OK)
        ESP_LOGE(TAG, "Failed to register RC522 events: %s", esp_err_to_name(err));
    
    
    err = rc522_start(scanner);
    if(err != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to start RC522 scanner: %s", esp_err_to_name(err));
        rc522_destroy(scanner);
    }
        
}
```

now as you can see its just your same example code nothing new or fancy, but i just wanna initially check if the module is even present or not, and if not, then just do other things and forget about it. 

here is the log for you

```
 (208) cpu_start: App cpu up.
I (236) cpu_start: Pro cpu start user code
I (236) cpu_start: cpu freq: 160000000
I (236) cpu_start: Application information:
I (241) cpu_start: Project name:     RFID_RC522
I (246) cpu_start: App version:      1
I (250) cpu_start: Compile time:     Sep  7 2022 01:12:13
I (257) cpu_start: ELF file SHA256:  4ed91e1f5ce277fd...
I (263) cpu_start: ESP-IDF:          v4.4.1
I (268) heap_init: Initializing. RAM available for dynamic allocation:
I (275) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (281) heap_init: At 3FFB2DF8 len 0002D208 (180 KiB): DRAM
I (287) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (293) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (300) heap_init: At 4008DEC0 len 00012140 (72 KiB): IRAM
I (307) spi_flash: detected chip: generic
I (311) spi_flash: flash io: dio
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (327) rc522: Read/write test failed
I (327) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (427) rc522: write: Unknown transport
E (427) rc522: Failed to write data (err: ESP_ERR_INVALID_STATE)
E (427) rc522: write: Unknown transport
E (427) rc522: Failed to write data (err: ESP_ERR_INVALID_STATE)
E (437) rc522: read: Unknown transport
E (437) rc522: Failed to read data (err: ESP_ERR_INVALID_STATE)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d557f  PS      : 0x00060530  A0      : 0x800d5703  A1      : 0x3ffb7ea0  
0x400d557f: rc522_read at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:92
 (inlined by) rc522_clear_bitmask at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:105
 (inlined by) rc522_card_write at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:337

A2      : 0x3ffaff6c  A3      : 0x0000000c  A4      : 0x00000030  A5      : 0xfffffff7
A6      : 0x3ffb7f10  A7      : 0xffffff80  A8      : 0x800d557f  A9      : 0x3ffb7e80
A10     : 0x00000000  A11     : 0x000001b5  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x3ffaff6c  A15     : 0x00000002  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa

Backtrace:0x400d557c:0x3ffb7ea00x400d5700:0x3ffb7ee0 0x400d5759:0x3ffb7f10 0x400d57e7:0x3ffb7f40 0x40089a4d:0x3ffb7f70
0x400d557c: rc522_read at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:91
 (inlined by) rc522_clear_bitmask at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:105
 (inlined by) rc522_card_write at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:337

0x400d5700: rc522_request at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:400

0x400d5759: rc522_get_tag at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:433

0x400d57e7: rc522_task at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:661

0x40089a4d: vPortTaskWrapper at C:/ESPIDF32/esp-idf/components/freertos/port/xtensa/port.c:131
```

i would definitely personally take interest in this and try and fix this, but before i do that, i thought it is better to point it out to you so you can shed some light on this.
Thanks a ton in advance 
Cheers!

abobija, 2022-09-08:
Hi @parthbhat13, thanks for feedback.
Can you please try with pinout wiring like in the example code:

```
    .spi.miso_gpio = 25,
    .spi.mosi_gpio = 23,
    .spi.sck_gpio = 19,
    .spi.sda_gpio = 22,
```

And let me know if it is working.

parthbhat13, 2022-09-11:
Hi @abobija, apologies for the delay.
I tried what you asked me to do to change the pins, but still the result is the exact same 

```
I (208) cpu_start: App cpu up.
I (236) cpu_start: Pro cpu start user code
I (236) cpu_start: cpu freq: 160000000
I (236) cpu_start: Application information:
I (241) cpu_start: Project name:     RFID_RC522
I (246) cpu_start: App version:      1
I (250) cpu_start: Compile time:     Sep  7 2022 01:12:13
I (257) cpu_start: ELF file SHA256:  e618b131fe373aea...
I (263) cpu_start: ESP-IDF:          v4.4.1
I (268) heap_init: Initializing. RAM available for dynamic allocation:
I (275) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (281) heap_init: At 3FFB2DF8 len 0002D208 (180 KiB): DRAM
I (287) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (293) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (300) heap_init: At 4008DEC0 len 00012140 (72 KiB): IRAM
I (307) spi_flash: detected chip: generic
I (311) spi_flash: flash io: dio
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (327) rc522: Read/write test failed
I (327) gpio: GPIO[22]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (427) rc522: write: Unknown transport
E (427) rc522: Failed to write data (err: ESP_ERR_INVALID_STATE)
E (427) rc522: write: Unknown transport
E (427) rc522: Failed to write data (err: ESP_ERR_INVALID_STATE)
E (437) rc522: read: Unknown transport
E (437) rc522: Failed to read data (err: ESP_ERR_INVALID_STATE)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d557f  PS      : 0x00060530  A0      : 0x800d5703  A1      : 0x3ffb7ea0
0x400d557f: rc522_read at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:92
 (inlined by) rc522_clear_bitmask at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:105
 (inlined by) rc522_card_write at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:337

A2      : 0x3ffaff6c  A3      : 0x0000000c  A4      : 0x00000030  A5      : 0xfffffff7
A6      : 0x3ffb7f10  A7      : 0xffffff80  A8      : 0x800d557f  A9      : 0x3ffb7e80
A10     : 0x00000000  A11     : 0x000001b5  A12     : 0x00000001  A13     : 0x00000000
A14     : 0x3ffaff6c  A15     : 0x00000002  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffa

Backtrace:0x400d557c:0x3ffb7ea00x400d5700:0x3ffb7ee0 0x400d5759:0x3ffb7f10 0x400d57e7:0x3ffb7f40 0x40089a4d:0x3ffb7f70
0x400d557c: rc522_read at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:91
 (inlined by) rc522_clear_bitmask at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:105
 (inlined by) rc522_card_write at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:337

0x400d5700: rc522_request at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:400

0x400d5759: rc522_get_tag at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:433

0x400d57e7: rc522_task at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:661

0x40089a4d: vPortTaskWrapper at C:/ESPIDF32/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: e618b131fe373aea

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:14788
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?
```

the next option I can think of is to use the i2c, and see how it works, if not that, then I have the irq pin which I am not using, so maybe I can short the IRQ pin on the MFRC board, and on the esp32 can just simply read the input if the pin is high or low, that's all. that ways we can know if the rfid module is present or not. 

I will update you with the i2c and let you know. 
Thanks for the reply
Cheers!

parthbhat13, 2022-09-11:
Hi @abobija , 

so below are the results for the i2c

```
I (29) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (29) boot: compile time 13:33:16
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b2e4h ( 45796) map
I (128) esp_image: segment 1: paddr=0001b30c vaddr=3ffb0000 size=024d0h (  9424) load
I (132) esp_image: segment 2: paddr=0001d7e4 vaddr=40080000 size=02834h ( 10292) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1b32ch (111404) map
I (182) esp_image: segment 4: paddr=0003b354 vaddr=40082834 size=0b68ch ( 46732) load
I (202) esp_image: segment 5: paddr=000469e8 vaddr=50000000 size=00010h (    16) load
I (209) boot: Loaded app from partition at offset 0x10000
I (209) boot: Disabling RNG early entropy source...
I (222) cpu_start: Pro cpu up.
I (222) cpu_start: Starting app cpu, entry point is 0x400811b4
0x400811b4: call_start_cpu1 at C:/ESPIDF32/esp-idf/components/esp_system/port/cpu_start.c:160

I (208) cpu_start: App cpu up.
I (236) cpu_start: Pro cpu start user code
I (236) cpu_start: cpu freq: 160000000
I (236) cpu_start: Application information:
I (241) cpu_start: Project name:     RFID_RC522
I (246) cpu_start: App version:      1
I (250) cpu_start: Compile time:     Sep 11 2022 13:32:53
I (257) cpu_start: ELF file SHA256:  ea42f6b735837b9e...
I (263) cpu_start: ESP-IDF:          v4.4.1
I (268) heap_init: Initializing. RAM available for dynamic allocation:
I (275) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (281) heap_init: At 3FFB2DF8 len 0002D208 (180 KiB): DRAM
I (287) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (293) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (300) heap_init: At 4008DEC0 len 00012140 (72 KiB): IRAM
I (307) spi_flash: detected chip: generic
I (311) spi_flash: flash io: dio
I (316) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (327) rc522: Failed to write data (err: ESP_FAIL)
E (337) rc522: Read/write test failed
E (337) rc522-demo: Failed to start RC522 scanner: ESP_FAIL
W (347) rc522: destroy_transport: Unknown transport

assert failed: spinlock_acquire spinlock.h:123 ((result == SPINLOCK_FREE) == (lock->count == 0))

Backtrace:0x400819ea:0x3ffb58900x40086d7d:0x3ffb58b0 0x4008cba5:0x3ffb58d0 0x40089c2d:0x3ffb59f0 0x40087c41:0x3ffb5a10 0x40087db8:0x3ffb5a50 0x400ea9ec:0x3ffb5a70 0x400d58f9:0x3ffb5aa0 0x400d5081:0x3ffb5ac0 0x400eb168:0x3ffb5b10 0x40089a4d:0x3ffb5b30
0x400819ea: panic_abort at C:/ESPIDF32/esp-idf/components/esp_system/panic.c:402

0x40086d7d: esp_system_abort at C:/ESPIDF32/esp-idf/components/esp_system/esp_system.c:128

0x4008cba5: __assert_func at C:/ESPIDF32/esp-idf/components/newlib/assert.c:85

0x40089c2d: spinlock_acquire at C:/ESPIDF32/esp-idf/components/esp_hw_support/include/soc/spinlock.h:123
 (inlined by) xPortEnterCriticalTimeout at C:/ESPIDF32/esp-idf/components/freertos/port/xtensa/port.c:288

0x40087c41: vPortEnterCritical at C:/ESPIDF32/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578
 (inlined by) xQueueSemaphoreTake at C:/ESPIDF32/esp-idf/components/freertos/queue.c:1563

0x40087db8: xQueueTakeMutexRecursive at C:/ESPIDF32/esp-idf/components/freertos/queue.c:731

0x400ea9ec: esp_event_loop_delete at C:/ESPIDF32/esp-idf/components/esp_event/esp_event.c:649

0x400d58f9: rc522_destroy at E:\ESP32_Projects\RFID_RC522\build/../components/RC522_RFID/RC522.c:561

0x400d5081: app_main at E:\ESP32_Projects\RFID_RC522\build/../main/main.c:58 (discriminator 3)

0x400eb168: main_task at C:/ESPIDF32/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x40089a4d: vPortTaskWrapper at C:/ESPIDF32/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: ea42f6b735837b9e

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6608
load:0x40078000,len:14788
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
```

seems like for now I might work over with the IRQ spare pin logic. if you can find a way to solve this, or even if I can get into this would update you. 

thanks again

closed 2022-10-03 by abobija