abobija/esp-idf-rc522 #13: Unexpected primary token before '.'

proffalken, 2022-09-21:
Hi!

I'm trying to use this library with a Heltec WiFi LoRa 32 v2 board that also has an SPI screen attached.

When compiling the code, I'm seeing the following output from Platformio:

```bash
src/main.cpp: In function 'void app_main()':
src/main.cpp:95:9: error: expected primary-expression before '.' token
         .spi.host = VSPI_HOST,
         ^
src/main.cpp:96:9: error: expected primary-expression before '.' token
         .spi.miso_gpio = 25,
         ^
src/main.cpp:97:9: error: expected primary-expression before '.' token
         .spi.mosi_gpio = 23,
         ^
src/main.cpp:98:9: error: expected primary-expression before '.' token
         .spi.sck_gpio = 19,
         ^
src/main.cpp:99:9: error: expected primary-expression before '.' token
         .spi.sda_gpio = 22,
         ^
src/main.cpp:102:19: error: 'config' was not declared in this scope
     rc522_create(&config, &scanner);
                   ^~~~~~
src/main.cpp:102:19: note: suggested alternative: 'confstr'
     rc522_create(&config, &scanner);
                   ^~~~~~
                   confstr
src/main.cpp:94:20: warning: unused variable 'rc522_config' [-Wunused-variable]
     rc522_config_t rc522_config = {
                    ^~~~~~~~~~~~
```

The full application code is as follows:

```cpp
/*******************************************************************************
 * 
 * ttn-esp32 - The Things Network device library for ESP-IDF / SX127x
 * 
 * Copyright (c) 2018 Manuel Bleichenbacher
 * 
 * Licensed under MIT License
 * https://opensource.org/licenses/MIT
 *
 * Sample program showing how to send and receive messages.
 *******************************************************************************/

#include "freertos/FreeRTOS.h"
#include "esp_event.h"
#include "driver/gpio.h"
#include "nvs_flash.h"

#include "TheThingsNetwork.h"

// NOTE:
// The LoRaWAN frequency and the radio chip must be configured by running 'idf.py menuconfig'.
// Go to Components / The Things Network, select the appropriate values and save.

// Copy the below hex strings from the TTN console (Applications > Your application > End devices
// > Your device > Activation information)

// AppEUI (sometimes called JoinEUI)
const char *appEui = "0000000000000000";
// DevEUI
const char *devEui = "<DEVEUI>";
// AppKey
const char *appKey = "<APPKEY>";

// Pins and other resources
#define TTN_SPI_HOST      SPI2_HOST
#define TTN_SPI_DMA_CHAN  SPI_DMA_DISABLED
#define TTN_PIN_SPI_SCLK  5
#define TTN_PIN_SPI_MOSI  27
#define TTN_PIN_SPI_MISO  19
#define TTN_PIN_NSS       18
#define TTN_PIN_RXTX      TTN_NOT_CONNECTED
#define TTN_PIN_RST       14
#define TTN_PIN_DIO0      26
#define TTN_PIN_DIO1      35

// RFID Reader Setup
#include <esp_log.h>
#include <inttypes.h>
#include "rc522.h"
static const char* TAG = "rc522-demo";
static rc522_handle_t scanner;

static void rc522_handler(void* arg, esp_event_base_t base, int32_t event_id, void* event_data)
{
    rc522_event_data_t* data = (rc522_event_data_t*) event_data;

    switch(event_id) {
        case RC522_EVENT_TAG_SCANNED: {
                rc522_tag_t* tag = (rc522_tag_t*) data->ptr;
                ESP_LOGI(TAG, "Tag scanned (sn: %" PRIu64 ")", tag->serial_number);
            }
            break;
    }
}

static TheThingsNetwork ttn;

const unsigned TX_INTERVAL = 30;
static uint8_t msgData[] = "Hello, world";

void sendMessages(void* pvParameter)
{
    while (1) {
        printf("Sending message...\n");
        TTNResponseCode res = ttn.transmitMessage(msgData, sizeof(msgData) - 1);
        printf(res == kTTNSuccessfulTransmission ? "Message sent.\n" : "Transmission failed.\n");

        vTaskDelay(TX_INTERVAL * pdMS_TO_TICKS(1000));
    }
}

void messageReceived(const uint8_t* message, size_t length, ttn_port_t port)
{
    printf("Message of %d bytes received on port %d:", length, port);
    for (int i = 0; i < length; i++)
        printf(" %02x", message[i]);
    printf("\n");
}

extern "C" void app_main(void)
{
    // Configure the RFID Reader
    rc522_config_t rc522_config = {
        .spi.host = VSPI_HOST,
        .spi.miso_gpio = 25,
        .spi.mosi_gpio = 23,
        .spi.sck_gpio = 19,
        .spi.sda_gpio = 22,
    };

    rc522_create(&config, &scanner);
    rc522_register_events(scanner, RC522_EVENT_ANY, rc522_handler, NULL);
    rc522_start(scanner);
    esp_err_t err;
    // Initialize the GPIO ISR handler service
    err = gpio_install_isr_service(ESP_INTR_FLAG_IRAM);
    ESP_ERROR_CHECK(err);
    
    // Initialize the NVS (non-volatile storage) for saving and restoring the keys
    err = nvs_flash_init();
    ESP_ERROR_CHECK(err);

    // Initialize SPI bus
    spi_bus_config_t spi_bus_config;
    spi_bus_config.miso_io_num = TTN_PIN_SPI_MISO;
    spi_bus_config.mosi_io_num = TTN_PIN_SPI_MOSI;
    spi_bus_config.sclk_io_num = TTN_PIN_SPI_SCLK;
    spi_bus_config.quadwp_io_num = -1;
    spi_bus_config.quadhd_io_num = -1;
    spi_bus_config.max_transfer_sz = 0;
    err = spi_bus_initialize(TTN_SPI_HOST, &spi_bus_config, TTN_SPI_DMA_CHAN);
    ESP_ERROR_CHECK(err);

    // Configure the SX127x pins
    ttn.configurePins(TTN_SPI_HOST, TTN_PIN_NSS, TTN_PIN_RXTX, TTN_PIN_RST, TTN_PIN_DIO0, TTN_PIN_DIO1);

    // The below line can be commented after the first run as the data is saved in NVS
    ttn.provision(devEui, appEui, appKey);

    // Register callback for received messages
    ttn.onMessage(messageReceived);

//    ttn.setAdrEnabled(false);
//    ttn.setDataRate(kTTNDataRate_US915_SF7);
//    ttn.setMaxTxPower(14);

    printf("Joining...\n");
    if (ttn.join())
    {
        printf("Joined.\n");
        xTaskCreate(sendMessages, "send_messages", 1024 * 4, (void* )0, 3, nullptr);
    }
    else
    {
        printf("Join failed. Goodbye\n");
    }
}
```

Without the rc522 library, the code compiles and connects to The Things Network.

I want to trigger a message that sends the card UID to TTN when someone taps a card to the reader, like the one at https://www.youtube.com/watch?v=etzn2lIEOX4&ab_channel=1nformatica but using LoRaWAN instead of LoRa.

I'm well out of my depth on this now, so any help you can provide would be more than welcome! :D

abobija, 2022-09-22:
Hi @proffalken

It seems like Platformio's C++ compiler does not support [Designated Initializers](https://gcc.gnu.org/onlinedocs/gcc/Designated-Inits.html), so you can try next:

Replace this:

```c
rc522_config_t rc522_config = {
    .spi.host = VSPI_HOST,
    .spi.miso_gpio = 25,
    .spi.mosi_gpio = 23,
    .spi.sck_gpio = 19,
    .spi.sda_gpio = 22,
};
```

With this:

```c
rc522_config_t rc522_config;

rc522_config.spi.host = VSPI_HOST,;
rc522_config.spi.miso_gpio = 25;
rc522_config.spi.mosi_gpio = 23;
rc522_config.spi.sck_gpio = 19;
rc522_config.spi.sda_gpio = 22;
```

I do not have tested but I think it should work. Let me know if it is working.

proffalken, 2022-09-23:
Thanks, that's definitely made some progress, I now get the following error:

```
Compiling .pio/build/heltec_wifi_lora_32_V2/esp-tls/esp_tls_error_capture.o
components/esp-idf-rc522/rc522.c: In function 'rc522_i2c_send':
components/esp-idf-rc522/rc522.c:583:12: error: implicit declaration of function 'i2c_master_write_to_device'; did you mean 'i2c_master_write_byte'? [-Werror=implicit-function-declaration]
     return i2c_master_write_to_device(rc522->config->i2c.port, RC522_I2C_ADDRESS, buffer, length, rc522->config->i2c.rw_timeout_ms / portTICK_PERIOD_MS);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~
            i2c_master_write_byte
Compiling .pio/build/heltec_wifi_lora_32_V2/esp-tls/esp_tls_mbedtls.o
components/esp-idf-rc522/rc522.c: In function 'rc522_i2c_receive':
components/esp-idf-rc522/rc522.c:588:12: error: implicit declaration of function 'i2c_master_write_read_device'; did you mean 'i2c_master_write_byte'? [-Werror=implicit-function-declaration]
     return i2c_master_write_read_device(rc522->config->i2c.port, RC522_I2C_ADDRESS, &addr, 1, buffer, length, rc522->config->i2c.rw_timeout_ms / portTICK_PERIOD_MS);
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
            i2c_master_write_byte
Compiling .pio/build/heltec_wifi_lora_32_V2/esp32/cache_err_int.o
Archiving .pio/build/heltec_wifi_lora_32_V2/esp-idf/efuse/libefuse.a
Indexing .pio/build/heltec_wifi_lora_32_V2/esp-idf/efuse/libefuse.a
Compiling .pio/build/heltec_wifi_lora_32_V2/esp32/cache_sram_mmu.o
Compiling .pio/build/heltec_wifi_lora_32_V2/esp32/clk.o
cc1: some warnings being treated as errors
Compiling .pio/build/heltec_wifi_lora_32_V2/esp32/crosscore_int.o
Compiling .pio/build/heltec_wifi_lora_32_V2/esp32/dport_access.o
*** [.pio/build/heltec_wifi_lora_32_V2/components/esp-idf-rc522/rc522.o] Error 1
```

abobija, 2022-09-25:
Ok, that looks like Platformio have not merged new version of ESP-IDF where that i2c functions are implemented. So you have two options: use esp-idf to make your application or wait for Platformio to merge new version of esp-idf.

abobija, 2022-09-25:
I will close this issue since this is related to Platformio platform, but feel free to write here more comments if you have some doubts.

closed 2022-09-25 by abobija

proffalken, 2022-09-25:
> Ok, that looks like Platformio have not merged new version of ESP-IDF where that i2c functions are implemented. So you have two options: use esp-idf to make your application or wait for Platformio to merge new version of esp-idf.

@abobija thanks - do you know which version of espidf I need to be running for this to work?

No worries if not, just saves me digging through the changelings!