abobija/esp-idf-rc522 #38: Please, add a ".clang-format" file

supcik, 2024-05-24:
I would be happy to work on some TODO in this project (in particular the new I2C library for IDF >= 5.2.0). Could you please provide a ".clang-format" file with your preferences for the code formatting?

abobija, 2024-05-25:
Hi @supcik,

Thank you for your interest in enhancing the library. I don't have a .clang-format file for this project, but feel free to create one. You can find the formatting rules [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/contribute/style-guide.html#c-code-formatting).

supcik, 2024-05-27:
Thank you for your reply. I wrote a `.clang-format` file (hopefully) compatible with your style, and I reformatted all source files. I hope that you are happy with the result : https://github.com/supcik/esp-idf-rc522/tree/reformat

I also configured _github actions_ for the following tasks:

- Check the formatting rules
- Analyze code using static checker "cppcheck"
- Compile a small example for various combination of targets and IDF versions

Note, I had to replace `VSPI` by `SPI3_HOST` in the example to compile on esp32-s3.

If you agree with this contribution, I'll be happy to make a pull-request.

abobija, 2024-05-27:
Thank you so much for the effort. It looks good, feel free to open PR and I will check it on my side.

PR #44 (2024-09-08): .clang-format file and check workflow
- Added `.clang-format` file
- Added workflow for checking if PR follows clang-format rules

Closes #38
commit 6ee15a77 (2024-09-08): .clang-format file and check workflow (#44)

- Added `.clang-format` file
- Added workflow for checking if PR follows clang-format rules

Closes #38
closed 2024-09-08 by abobija