abobija/esp-idf-rc522 #71: Fifo length mismatch

Zkall123, 2024-10-17:
Hello, I tried your basic example from the main branch and encountered some errors.

Here is the log output:
```
I (11773) rc522-basic-example: rc522_spi_create
I (11773) rc522-basic-example: rc522_driver_install
I (11783) rc522-basic-example: rc522_create
I (11783) rc522-basic-example: rc522_register_events
I (11793) rc522-basic-example: rc522_start
E (11813) rc522: rc522_pcd_rw_test(292): FIFO length missmatch after write
E (11813) rc522: rc522_start(86): rw test failed
```

Could you please help me resolve this issue?

abobija, 2024-10-17:
> the esp32 is the lilligo esp32-s3 eth lite

^ hmm, that could actually be the issue.

I’ve only tested with a regular ESP-32, not with the S3.
I think I have an S3R8 somewhere, so I’ll test with it soon.

Thank you for the feedback.

abobija, 2024-10-17:
I've tested with S3 and it works.

These GPIOs in basic example are for regular esp32:
https://github.com/abobija/esp-idf-rc522/blob/2d0bfb423148c5d6e8801d8f3d52aacd30e3d7b3/examples/basic/main/basic.c#L8-L11

so for mine S3, I've defined them like this:
```c
#define RC522_SPI_BUS_GPIO_MISO    (13)
#define RC522_SPI_BUS_GPIO_MOSI    (12)
#define RC522_SPI_BUS_GPIO_SCLK    (11)
#define RC522_SPI_SCANNER_GPIO_SDA (10)
```

and instead of `VSPI_HOST` on line `15`:
https://github.com/abobija/esp-idf-rc522/blob/2d0bfb423148c5d6e8801d8f3d52aacd30e3d7b3/examples/basic/main/basic.c#L15

I've used `SPI3_HOST`, since `VSPI_HOST` is not defined for S3:
```c
    .host_id = SPI3_HOST,
```

If you already done everything as me, please check again if you have correct wiring between ESP32 and RC522.

Zk4ll, 2024-10-18:
That's strange. Could the issue be that I am using LAN over SPI? The pins you defined are the same ones my board uses for the LAN port.

abobija, 2024-10-18:
Yes, that could be the issue. If you are using LAN over SPI that means SPI bus is already initialized and don't need to be reinitialized for RC522.

To attach RC522 to already initialized SPI bus, just ommit `bus_config` in driver configuration.

So this lines:
https://github.com/abobija/esp-idf-rc522/blob/2b6c968acd56eed5f16fbdd7f6e839dea501b570/examples/basic/main/basic.c#L14-L25

Should look like this (without `bus_config`):
```c
static rc522_spi_config_t driver_config = {
    .host_id = SPI3_HOST,
    .dev_config = {
        .spics_io_num = RC522_SPI_SCANNER_GPIO_SDA,
    },
    .rst_io_num = RC522_SCANNER_GPIO_RST,
};
```

That will attach RC522 to bus on `SPI3_HOST`. So just make sure that your LAN over SPI is already initialized, and use `SPI3_HOST`, before you call functions for initialization of RC522.

Let me know if it works :)

Zkall123, 2024-10-18:
It seems that the SPI pins for the LAN bus on this board weren't exposed by the manufacturer. Is there any way to add an additional SPI bus?

abobija, 2024-10-18:
I'm not sure what you mean by "exposed," but you don't need an additional SPI bus. You can connect the RC522 to the same MISO, MOSI, and SCK pins that the LAN bus uses. Just assign a free GPIO pin for the RC522's CS (SDA) pin.

For example, if your LAN over SPI is using:

MISO: GPIO 11
MOSI: GPIO 12
SCK: GPIO 10
CS: GPIO 9

Then for the RC522, use the same three SPI pins and pick an unused GPIO for the RC522's CS pin:

MISO -> GPIO 11 (same as LAN)
MOSI -> GPIO 12 (same as LAN)
SCK -> GPIO 10 (same as LAN)

But if this doesn't work for some reason, try to use different host than SPI3.

Zkall123, 2024-10-18:
I'm sorry, I meant that the LAN port is on the development board, and the SPI bus is not connected to the jumpers, so I don't have access to this SPI bus.

Zkall123, 2024-10-18:
Okay, thanks! I'll try it when I'm home. Are there any other steps besides using SPI2_host instead of SPI3_host?

abobija, 2024-10-18:
That should be enough. Just make sure to specify SPI2_HOST and correctly assign the MISO, MOSI, SCK, and CS pins in your code for the RC522. As long as the pins you choose are available, everything should work fine.

Zkall123, 2024-10-20:
I tried it again and got the same error. I also tried using spi1_host, but that didn't work either. I checked all the pin connections, and they seem fine. Could it be related to the DMA channel?

abobija, 2024-11-06:
I'm glad you managed to solve it.

closed 2024-11-06 by abobija