abobija/esp-idf-rc522 #76: Error reading MIFARE block address

kungfupizza, 2025-01-02:
Hello! I have been using your library. It has worked out good until reading the UID but now that I am trying to read from the memory, I have the following issue. Here are the prints,

```
I (15696) rfid-man: Tag scanned (sn: 585608387)
E (15724) rfid-man: Auth error = 63013
I (15724) rfid-man: Reading data from the block 4
E (15750) rc522: rc522_mifare_read(216): F528
```

I am using your code as in [example](https://github.com/abobija/esp-idf-rc522/blob/main/examples/read_write/main/read_write.c) with default keys being used.

I have also toyed with changing default KeyA to 0x00000000 but hasn't worked out. Please help!

Attaching information screen of the card from NFC tools.
[image]

abobija, 2025-03-16:
Closed due to inactivity

closed 2025-03-16 by abobija

enes-vural, 2025-04-12:
```
  rc522_mifare_key_t key = {
      .value = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF},
      .type = RC522_MIFARE_KEY_B,
  };

```
I was using NFC Tools PRO to change NDEF data inside the card. Setting the type to RC522_MIFARE_KEY_B solved my issue.