abobija/esp-idf-rc522 #77: Variable may be used uninitialized warning

kisbogdan-kolos, 2025-02-14:
```
components/rc522/src/rc522_picc.c: In function 'rc522_picc_select':
components/rc522/src/rc522_picc.c:551:34: error: 'tx_last_bits' may be used uninitialized [-Werror=maybe-uninitialized]
  551 |         if (response_length != 3 || tx_last_bits != 0) { // SAK must be exactly 24 bits (1 byte + CRC_A).
      |             ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
components/rc522/src/rc522_picc.c:305:13: note: 'tx_last_bits' was declared here
  305 |     uint8_t tx_last_bits; // Used in BitFramingReg. The number of valid bits in the last transmitted byte.
      |             ^~~~~~~~~~~~
```

The variable is initialized in every scenario, just need to add `#pragma GCC diagnostic ignored "-Wmaybe-uninitialized"` before line 305.

abobija, 2025-03-07:
On which esp-idf version and esp chip this happens?

abobija, 2025-03-16:
Closed due to inactivity

closed 2025-03-16 by abobija