abobija/esp-idf-rc522 #82: RC522 SPI design and the ESP32 limit of 3 CS pins for SPI3

kanflo, 2025-03-12:
I have been struggling with adding 4x RC522s and just learned about the ESP32 limit of 3 CS pins. For my part this calls for an update of `rc522_spi_config_t` where a "soft" CS pin can be set and setting `spics_io_num` to -1 (and a change for pulling these pins, if configured, in `rc522_spi_send/receive`). Could this be of PR interest?

Regarding the SPI design, I like the fact that only the first RC522 needs an `spi_bus_config_t` while subsequent RC522s simply can say "I'm on the same bus". I was a bit surprised though that every RC522 gets a separate thread rather than a single thread per SPI bus, alternating between the RC522s on that bus. I guess most users only have one single reader but for my 4x use case I feel RAM usage could need an improvement. I'll see if this ends up in a PR too.

kanflo, 2025-03-29:
I am currently looking at modifying the `rc522_spi_config_t` struct to allow for a "soft CS pin" with callbacks added to `dev_config`.

closed 2025-03-29 by kanflo

kanflo, 2025-06-24:
I just opened a PR on this, see https://github.com/abobija/esp-idf-rc522/pull/89

gid204, 2025-07-16:
Hey @kanflo sorry this probably isn't the best place to ask for help, but I'm not sure where else to message for now.

I'm trying to use your fork (PR) to run the read_write example and am getting an error on boot:
```
I (332) main_task: Calling app_main()
E (352) rc522: rc522_pcd_rw_test(292): FIFO length missmatch after write
E (352) rc522: rc522_start(86): rw test failed
I (352) main_task: Returned from app_main()

```

This is my spi config:
```
#define RC522_SPI_BUS_GPIO_MISO    (25)
#define RC522_SPI_BUS_GPIO_MOSI    (23)
#define RC522_SPI_BUS_GPIO_SCLK    (19)
#define RC522_SPI_SCANNER_GPIO_SDA (22)
#define RC522_SCANNER_GPIO_RST     (-1) // soft-reset

static rc522_spi_config_t driver_config = {
    .host_id = SPI3_HOST,
    .bus_config = &(spi_bus_config_t){
        .miso_io_num = RC522_SPI_BUS_GPIO_MISO,
        .mosi_io_num = RC522_SPI_BUS_GPIO_MOSI,
        .sclk_io_num = RC522_SPI_BUS_GPIO_SCLK,
    },
    .dev_config = {
        .spics_io_num = -1,
    },
    .rst_io_num = RC522_SCANNER_GPIO_RST,
    .ncs_io_num = RC522_SPI_SCANNER_GPIO_SDA
};
```

When I run the latest release of the component it works fine.

Am I missing something?
Cheers!

kanflo, 2025-07-16:
Hi @gid204 That's strange, I tried your exact config with one single scanner and if worked flawlessly. I guess we should move this discussion to the PR though. I tried setting an incorrect nCS for the reader but did not make it past `rc522_pcd_reset` in `rc522_start`, that indicates that your nCS definition is ok.

Some questions come to mind:
1. Are you using one scanner only?
2. What happens if you move `RC522_SPI_SCANNER_GPIO_SDA` to `.spics_io_num` and remove `.ncs_io_num`?
3. Any other logs? I get the following:
```I (5792) spi_master: *** spi_bus_add_device
W (5792) spi_bus_lock: *** try_acquire_free_dev cs not required
I (5802) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (5852) rc522: PCD (fw=v2.0) initialized```

gid204, 2025-07-16:
Moving this conversation to the PR: https://github.com/abobija/esp-idf-rc522/pull/89