abobija/esp-idf-rc522 #83: Reading/writing tags outside of the RC522_EVENT_PICC_STATE_CHANGED event loop

kanflo, 2025-03-13:
The read/write examples all perform these operations inside of the state change event loop which I guess is called within the context of the `rc522_task` (please correct me if I'm wrong). As there are no mutexes for accessing the driver "from the outside", eg. on `rc522_nxp_write`, I guess this is not supported. Or have I missed something?

abobija, 2025-03-14:
There is `task_mutex` inside of config

https://github.com/abobija/esp-idf-rc522/blob/main/include/rc522_types.h#L40

which is taken on every poll inside of `rc522_task`

https://github.com/abobija/esp-idf-rc522/blob/main/src/rc522.c#L238-L241

which means that `rc522_task` will wait if you have taken mutex outside.

Can this help you maybe?

kanflo, 2025-03-29:
Thanks! That could help of course but I would not mind keeping this semaphore inside the rc522 module. I might raise a PR with this functionality as I think it would improve "usability" of this module.

closed 2025-03-29 by kanflo