abobija/esp-idf-rc522 #93: E (979115) rc522: rc522_picc_send(86):

usmanrazasaif, 2025-08-09:
The FW goes into a loop and does not detect tags anymore. This issue is very random.

abobija, 2025-08-09:
Let me know when/if you find exact reproduce steps for this issue.

usmanrazasaif, 2025-12-04:
It occurs very randomly and quite often on some hardware. Here is the complete log. 

W (1460201) rc522: rc522_picc_receive(125): fifo empty (irq=0xFF)
E (1460202) rc522: rc522_picc_transceive(206): F52F
W (1460204) rc522: rc522_task(282): select failed (err=F52F)
E (1460298) rc522: rc522_picc_send(86):
E (1460386) rc522: rc522_picc_send(86):
E (1460474) rc522: rc522_picc_send(86):
E (1460562) rc522: rc522_picc_send(86):
E (1460650) rc522: rc522_picc_send(86):
E (1460738) rc522: rc522_picc_send(86):
E (1460826) rc522: rc522_picc_send(86):
E (1460913) rc522: rc522_picc_send(86):
E (1461002) rc522: rc522_picc_send(86):

I guess a soft reset mechanism is required to handle this case. The driver stops working after it